sumotoy/SSD_13XX #2: SSD1531 does not work at 96Mhz

cosmikwolf, 2016-08-26:
SSD1531 does not work at 96Mhz

Either shows garbage pixels or nothing at all.

cosmikwolf, 2016-08-26:
I think this might be because the SPI clock speed is too fast at 96Mhz. 

When running the benchmark example sketch, I took pictures of a scope on the SCLK line:
at 72Mhz:
[image]

and then at 96mhz:
[image]

The SSD1351 datasheet says that the minimum time between clock signals is 50ns, as you can see the divisions between the lines on the scope is 50ns, and the 96mhz SPI clock is under 50ns, while the 72Mhz SPI clock seems to be just at 50ns.

cosmikwolf, 2016-08-26:
it also seems that if I change the speed in _cpuCommons.h to 23999999 it just jumps back down to 16mhz, which seems like it might be too slow?

cosmikwolf, 2016-08-26:
at 72mhz CPU, SPI clock goes at clockdiv4, which is 18mhz, which seems to be just what the SSD1351 likes....

cosmikwolf, 2016-08-26:
It seems to work at 8mhz SPI Clock, at 96Mhz CPU clock

sumotoy, 2016-08-27:
clockdiv4 have nothing to do, Teensy use SPI transactions so the speed is directly handled in Mhz.
Try open _includes/_cpuCommons and change the SPI speed for Teensy, so try set 18000000.
The SPI speed in SPI transaction mode is handled like this: internally the value is corrected to match the legal values in relation to CPU speed.

I have 2 different SSD1531 based display here (chinese made), both works at all speed and I'm really sure that are SSD1531, I exposed MISO pin and writed a small code to identify it.
Are you sure there's any resistor,condenser in the display SPI lines? Or there's any other SPI device sharing same line (like SD card)?

cosmikwolf, 2016-08-27:
I understand that you can specify the frequency in megahertz, but that is just the max frequency upper limit.  when you view the SCLK lines on an oscilloscope, You can see that the teensy will only send spi clock at certain frequencies, like 8mhz, 12mhz, 16mhz, 18mhz, 24mhz. At 96mhz fclock speed though, it will not send a 18mhz SPI clock. 

The board I have does have other SPI devices such as an sd card, but these are all connected properly and do not interfere with other spi mode0 devices. There is a 50 ohm source termination resistor inline to match line resistance and to improve the SPI clock waveform and reduce clock reflections. 

I'm just trying to help you find the source of this bug. 

Are the SSD1351 displays you are using that work 128x128? The ones I am using are 128x96. They are attached to boards I designed based on the Solomon datasheet. 

As I said before though, it seems to work fine when you limit the SPI clock to 12mhz. Perhaps the 128x96 display does not like 16mhz. I tried changing the internal oscillator offset (I am not at my computer right now so I don't have the hex address in front of me)

Thank you for your work!