sumotoy/SSD_13XX #8: Wemos D1 mini pro

weigu1, 2017-05-12:
Hi,
I used the SSD1331 from adafruit with a wemos d1 mini pro (ESP8266, CS on D3, DC on D0, RST with 4,7k on 5V)
Text is not readable (from tft.print("HHH") I see only the middle line of the H)
tft.drawLine(20, 20, 40, 40, BLUE) is visible, but  tft.drawCircle and tft.drawRoundRect don't show anything.
Thank you for the lib and for your help.
Greetings

unorthodox-engineers, 2017-06-07:
I think I'm seeing the same problem on a generic ESP-12E module I've got wired up on the breadboard to a SD1331 display from eBay. Any text at setTextScale(1) is missing every second line, though setTextScale(2) and up works OK. I tried several fonts and they all did the same thing. fillRect() works, but drawPixel() does not. lines do work, circles don't.

It feels like something is wrong in the single-pixel draw routines. Anything that uses fast H or V lines (rects, lines, big fonts) to draw many pixels at once seem to be OK.

It took me a while to figure out how to wire it up. Seems like the HSPI-CS pin (GPIO15) is being used as CS no matter what I say in the code.

Otherwise it's very fast, the colors are nice, and I wish it was working. I may have to dig into the code myself and see what's up. I suspect it's some bitwise math the ESP8266 compiler is doing it's own unique way, or perhaps some issue with very short SPI transfers.

unorthodox-engineers, 2017-06-07:
Of course, ten minutes after posting it started working. (then stopped, then started again) I now think it's to do with the DC pin choice (The behavior was the same when I deliberately set the DC pin wrong.) I haven't fully deciphered the logic, but pins GPIO5 and GPIO4 seem to work as DC sometimes, but GPIO16 does not.