sumotoy/SSD_13XX #9: Can't drawIcon an icon which width/height equal to the width/height of the OLED?

paperadio, 2017-06-13:
I am using SSD1331 OLED, it's a 96 x 64 OLED display. I want to print a 64 x 64 QR code on display, but I can't see anything on display. I looked up source code  in SSD_13XX.cpp, and found this in line 1921:
`if ((x + iWidth) * scale >= _width || (y + iHeight) * scale >= _height) return;//cannot be`
When drawIcon is called, it always return, because my icon is:
`static const tIcon qrcode1 = { image_data_qrcode1, 64, 64, 512, RLE_no }; `
That is the reason in my case. so can't I draw a height 64 icon? why? thanks a lot.

atc1441, 2017-07-09:
I think you have to use height -1 so 63 because the pixels starting at 0 and not 1. 
With 64 it is to big for the display.

paperadio, 2018-06-27:
Thanks, I got it. I gave up the QR idea already, because My phone cannot recognize the QR  code image on the OLED successfully, maybe the resolution is too low to decode a QR code.

closed 2018-06-27 by paperadio