sumotoy/SSD_13XX #16: Custom Fonts

Betacarotin, 2018-02-05:
Hi! 
Thankls for sharing this!
I got a small Issue, how can I load in a custom font? I managed it with adafruit GFX due to a webconverter. Now I am quite unsure how to get these .c files for fonts. Hopefully someone can help me?

yashm, 2018-03-26:
@sumotoy any tips on how to make custom fonts?
Tried a bunch of settings on the "LCD Image Converter" but get compile errors.

Thanks!

phillip-toone, 2019-04-10:
Yes, I would also like some guidance on how to create or at least modify an existing font.  This [font creation tool](http://oleddisplay.squix.ch/) creates four different formats for .h header files but I can't seem to understand how to implement them into this library.  I have attempted to manually copy and paste the hex code from this tool into the .c files but this usually just causes the code to no longer compile.  It looks like there is a "LCD Image Converter" but I also can't figure out how to use this.  An example here would be really appreciated.  Thanks.

phillip-toone, 2019-04-15:
I have created a python script for converting fonts created with [this font creation tool](http://oleddisplay.squix.ch) into the font file format implemented with this library.  You can find this script on my fork of this library under the "_utility" folder.  Use this script as follows.

- Download  the desired font in the <3.0.0 file format (fontFile.h in this example)
- Execute the script: "convert_squix_font.py fontFile.h"
- The script will generate a file called fontFile.c for your use.

I have generated a pull request for this script to be implemented into the master branch of this library.

Keoni80, 2019-05-16:
can anyone tell me how to use the comand tft.setFont? thanks!!

liesenbergk, 2021-06-27:
> I have created a python script for converting fonts created with [this font creation tool](http://oleddisplay.squix.ch) into the font file format implemented with this library. You can find this [...]

Dear Phillip-Toone,

I am trying to use your font converter  and I get the following prompt message:

Microsoft Windows [Version 10.0.19042.1052]
(c) Microsoft Corporation. All rights reserved.

C:\Users\klaus\Music\Python>python convert_squix_font.py dialog16.f
Traceback (most recent call last):
  File "C:\Users\klaus\Music\Python\convert_squix_font.py", line 50, in <module>
    Hexes = data[indexStart:indexEnd]
TypeError: slice indices must be integers or None or have an __index__ method

C:\Users\klaus\Music\Python>

Would you have any hint for me how to succeed?

thanks

Klaus