sumotoy/SSD_13XX #18: Arduino UNO connection with SSD1331 0.95inch RGB OLED from Waveshare?

jguillod, 2018-05-13:
I own a SSD1331 0.95inch RGB OLED from Waveshare (resolution 96x64, RGB, 65K colorful, [specifications here](https://www.waveshare.com/wiki/0.95inch_RGB_OLED_(A)).
I have been able to find how to connect pins on an Arduino UNO and the use of [Adafruit library for Arduino](https://github.com/adafruit/Adafruit-SSD1331-OLED-Driver-Library-for-Arduino) allow me to display correctly (but very slow).
I cannot figure out how to connect it to use SSD_13XX. Compilation and upload of code to the UNO is okay but nothing is displayed (serial shows the appropriate text, so firmware is running). Is there some schema how to make the appropriate connection.
Thanks a lot.

divins, 2018-11-01:
Hi @jguillod did you finally managed to make it work? I'm at the same situation, adafruit library is working, and this one is printing on serial but nothing is shown at the display. Thanks!

jguillod, 2018-11-03:
No sorry, I had no time to find how to make this lib to work ! But still interested…

divins, 2018-11-03:
Hi again! I'm not sure it will work for you, but I got DC on the wrong pin (8 instead of 9 that usually goes on the examples) but it was not enough: I had to add the Reset pin when calling the library. In my case it is at pin 9, so the call is this way:

```
#include <SPI.h>
#include <SSD_13XX.h>

#define __CS1 	10
#define __DC 	8

SSD_13XX tft = SSD_13XX(__CS1, __DC, 9);
```

This worked! Hope it helps you too...