sumotoy/SSD_13XX #22: Terrific work! Great with ESP32

nikthefix, 2018-10-31:
This lib is great and working well with ESP32 Devkit V1. Just remember to pull up the OLED CS line otherwise during power up the line will float and the display may not init properly.

Tip:

I use 3 x SSD1331 oleds in parallel on the SPI bus with shared CS and tied high RST. This arrangement highlights any startup reliability issues as sometimes 1 will fail indicating a problem with my circuit layout or timing. If all 3 constantly work as expected then I feel I'm good to go. I do the same with I2C.

nicolaufg, 2018-12-31:
Anyone tried with a [TTGO T2](https://github.com/LilyGO/TTGO-T2-SSD1331-SD) ?? The screen is attached to the board and there's no way to get it to show anything using this library (others work, like olikraus UcgLib which is very slow, and the one shipped with the board which has less features). The examples in this lib do work i.e. graphictest sends to Serial the times for each benchmark and it looks good, but the screen shows nothing at all.

That's the pinout:
```
const uint8_t SCLK_OLED = 14; //SCLK
const uint8_t MOSI_OLED = 13; //MOSI (Master Output Slave Input)
const uint8_t CS_OLED = 15;
const uint8_t DC_OLED = 16; //OLED DC(Data/Command)
const uint8_t RST_OLED = 4; //OLED Reset
```
Based on the comment above, I added
`pinMode(CS_OLED , INPUT_PULLUP);`
to no avail.

I found this bug [here](https://www.esp32.com/viewtopic.php?t=439), but it is said to be fixed. All in all, anyone got it to work with that board?

nicolaufg, 2018-12-31:
Thanks for the quick answer... Yes, I'm using the right pins (work fine with other libs):
```
#define TFT_DC  16
#define TFT_CS 15
#define TFT_RST 4
SSD_13XX tft = SSD_13XX(TFT_CS, TFT_DC, TFT_RST);
```
Well, I can pass more pins to other libs, but this one fails because longer signatures are only available for teensies and such:
```
#if defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MK64FX512__) || defined(__MK66FX1M0__)
	SSD_13XX::SSD_13XX(const uint8_t cspin,const uint8_t dcpin,const uint8_t rstpin,const uint8_t mosi,const uint8_t sclk)
	{
		_cs   = cspin;
		_dc   = dcpin;
		_rst  = rstpin;
		_mosi = mosi;
		_sclk = sclk;
	}
#elif defined(__MKL26Z64__) //Teensy LC
	SSD_13XX::SSD_13XX(const uint8_t cspin,const uint8_t dcpin,const uint8_t rstpin,const uint8_t mosi,const uint8_t sclk)
	{
		_cs   = cspin;
		_dc   = dcpin;
		_rst  = rstpin;
		_mosi = mosi;
		_sclk = sclk;
		_useSPI1 = false;
		if ((_mosi == 0 || _mosi == 21) && (_sclk == 20)) _useSPI1 = true;
	}
#else //All the rest
	SSD_13XX::SSD_13XX(const uint8_t cspin,const uint8_t dcpin,const uint8_t rstpin)
	{
		_cs   = cspin;
		_dc   = dcpin;
		_rst  = rstpin;
	}
#endif

```

As a side note, mine is TTGO T2 v1.4 like the one at the top in my previous link.

nikthefix, 2018-12-31:
I've got version 1.2 I think but I'll check my code against yours when I
get back after the xmas break.

Nick

nikthefix, 2019-01-01:
Hi, me again.

No need to edit the libs just add a couple of lines to the Sumoto example.
Let's take the  Bubbles example:

#define __CS1 15
                     //EDIT THIS
#define __DC 16
                             //EDIT THIS

SSD_13XX tft = SSD_13XX(__CS1, __DC, 4);
       //EDIT THIS
SPIClass vspi(VSPI);
                           //ADD THIS

int16_t sine[SCALE+(SCALE/4)];
int16_t *cosi = &sine[SCALE/4];
int16_t angleX = 0, angleY = 0, angleZ = 0;
int16_t speedX = 0, speedY = 0, speedZ = 0;

int16_t xyz[3][NDOTS];
uint16_t col[NDOTS];
int pass = 0;

void initialize (void){
  uint16_t i;
  /* if you change the SCALE*1.25 back to SCALE, the program will
   * occassionally overrun the cosi array -- however this actually
   * produces some interesting effects as the BUBBLES LOSE CONTROL!!!!
   */
  for (i = 0; i < SCALE+(SCALE/4); i++)
    //sine[i] = (-SCALE/2) + (int)(sinf(PI2 * i / SCALE) * sinf(PI2 * i /
SCALE) * SCALE);
    sine[i] = (int)(sinf(PI2 * i / SCALE) * SCALE);
}

void setup()
{
  vspi.begin(14, -1, 13, 15); // SCK,MISO,MOSI,SS
                         //ADD THIS
  tft.begin();
  initialize();

That's it. The issue is that TTGO board is physically configured for 2
seperate SPI busses - one for SD and the other for OLED so without defining
VSPI as the oled driver it won't work.

Hope this helps.

Nick

nicolaufg, 2019-01-02:
Thanks a lot Nick, it works like charm! This lib does things FAST!!

Before this TTGO board I was using an Arduino Mega and logging data into a SD card. With this one, I'm uploading to ThingSpeak ([public data](https://thingspeak.com/channels/659221)) but I may also enable local logging. Some rough SD tests have worked so far without paying much attention to one or another SPI bus (other than providing the right pins during the init), but I'll bear in mind your comments if I get into troubles.

Have a great year!
Nico

nikthefix, 2019-01-02:
Yes this is a fast lib. I tried to manually optimise it to make it even
faster for displaying images and animations but quickly reached the limits
of the SSD1331 - it was tripping up and missing instructions - so I'm
settled on this as my 'go to' for this display.

Nick

Nick

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Yury-MonZon, 2020-10-10:
I've tried to replicate your bubbles example patches, but I still can't make it to display anything with this lib.
I'm having 1.6 board (TTGO T2 v1.6).
My pins are:
#define OLED_CS   15
#define OLED_RST  4
#define OLED_DC   16
#define OLED_MOSI 13
#define OLED_MISO 12
#define OLED_SCK  14

Any help would be appreciated. Thanks.