sumotoy/SSD_13XX #30: How to draw custom images

lorhof1, 2021-08-23:
Hello, I didn't find any guide on how to draw own images. How can I do this? Sorry if I'm missing something.

stylesuxx, 2021-08-23:
What do you mean by own images? You have an image file that you want to be drawn, or you want to draw from scratch? In any way, I would recommend taking a look at the examples dir, that should help you out.

lorhof1, 2021-08-23:
i want to draw an image to the display, i saw your images are in a format that is an array of the pixel colors. how can i convert a png to this format?

stylesuxx, 2021-08-23:
I am not the maintainer of this lib, but you can check the file - the header tells you everything you need to know https://github.com/sumotoy/SSD_13XX/blob/master/_images/eye/eye93.c

```
* preset name: Color R5G6B5
* data block size: 16 bit(s), uint16_t
* RLE compression enabled: no
* conversion type: Color, not_used not_used
* bits per pixel: 16
*
* preprocess:
*  main scan direction: top_to_bottom
*  line scan direction: forward
*  inverse: no
```

* 5 bits red
* 6 bits green
* 5 bits blue

Results in 16 bits per pixel. It is drawn line after line - so if you have an image with the size of 20x10 pixels, it will result in 200pixels resulting in 3200 bits.

I like to use a python PIL script to convert images to other image formats - but how you implement this is up to you.

lorhof1, 2021-08-23:
Where can i get this LCD-Image-converter and how do i set it up?